rosflight/airbourne_f4 #1: Working SPI between F4 and MPU6000

len0rd, 2017-08-04:
Just going to use this to keep track of some thoughts, if anyone wants to join in and help.

From the F2 std_periph documentation(pg 484), I think current implementation is a little shaky on steps 2&3:

```
2. Enable SCK, MOSI, MISO and NSS GPIO clocks using
RCC_AHB1PeriphClockCmd() function. In I2S mode, if an external clock source is
used then the I2S CKIN pin GPIO clock should also be enabled.
3. Peripherals alternate function:
    - Connect the pin to the desired peripherals' Alternate Function (AF) using GPIO_PinAFConfig() function - Configure the desired pin in alternate function by: GPIO_InitStruct->GPIO_Mode = GPIO_Mode_AF
    - Select the type, pull-up/pull-down and output speed via GPIO_PuPd, GPIO_OType and GPIO_Speed members
    - Call GPIO_Init() function In I2S mode, if an external clock source is used then the I2S CKI
```

Also, reading the f4 reference, I think we want NSS to be in SPI_NSS_Hard mode:

```
Hardware NSS management (SSM = 0)
Two configurations are possible depending on the NSS output configuration (SSOE bit in register SPI_CR2).
    – NSS output enabled (SSM = 0, SSOE = 1) This configuration is used only when the device operates in master mode. The NSS signal is driven low when the master starts the communication and is kept low until the SPI is disabled. 
```
This is the functionality required by the mpu6000. The SPI1 NSS line is connected to a pull-up resistor, so it needs to be driven low whenever we want to send/receive data.

len0rd, 2017-08-08:
Looks like most of the mpu6000 init stuff I was worried about is handled in mpu6000_spi constructor. At this point i think the biggest thing to check is the spi.transfer and spi.set_ss_low/high functions.

It may be necessary to adjust the baudrate_prescaler to adhere to mpu6000 1Mhz max

len0rd, 2017-08-10:
now might be a good time to try configuring the nss pin as od with no pull

len0rd, 2017-08-15:
Running out of ideas, it may be a problem with how I interact with the MPU. I hooked up the logic analyzer to SPI2, which comes out the Flex-IO port. SPI seems to work fine, the only strange thing is on this particular transfer, MOSI changes from inactive high to inactive low, despite setting it to have a PU resistor in the GPIO conf:

[image]

Not sure if this could cause issue, but its certainly unexpected. Also need to test the receive function

len0rd, 2017-08-18:
Think I finally have this working! At least I'm getting data back, so that's good. I kind of ended up rewriting all of the spi and mpu stuff though, so its going to be a few messy commits here...

len0rd, 2017-08-18:
blink and spi examples are working! (vcp would be totally broken even if I didnt remove the vcp.cpp files. Planning on fixing next) Currently mpu just spits back the raw int16_t data of all the sensors, which I think is fine since this is how the breezy mpu6050 driver works. Next is getting exti working, however I think working pwm may take priority over that.

superjax, 2017-08-19:
Nice!!

len0rd, 2017-08-22:
closing for now

closed 2017-08-22 by len0rd