rosflight/airbourne_f4 #2: Working PWM for ESC

len0rd, 2017-08-18:
Absolutely no ideas on this yet, but will probably combine ideas from Breezy's pwm driver and and [AQ32Plus pwm esc driver](https://github.com/jihlein/AQ32Plus/blob/master/src/drv/drv_pwmEsc.c)

superjax, 2017-08-19:
You might also want to take a look at my driver for the F1  https://github.com/superjax/airbourne/blob/master/src/pwm.cpp

len0rd, 2017-08-22:
Tested with logic analyzer and motor on all pwm outputs. Working!

Only little quirk is output3/TIM9 is very out of phase with the others, but this doesnt really matter:

[image]

len0rd, 2017-08-23:
tested on a couple servos and motors/escs at 50 and 490 Hz respectively. Also used logic analyzer to check for proper values. Looks good to go!

superjax, 2017-08-23:
Awesome!  Thanks Man!

closed 2017-09-25 by superjax