rosflight/airbourne_f4 #51: SBUS Incompatible with Futaba Receivers

ccackam, 2020-02-05:
We found that SBUS communication does not work with Futaba Receivers. We were able to track the issue down to the end byte. FrSky receivers use 0x00 as the end byte and this is how the code is setup currently. Futaba uses an of the following as end bytes to indicate a non corrupted packet: 0x04, 0x14, 0x24, 0x34.

We were able to get the code to work with a Futaba receiver by changing lines 46-50 of rc_sbus.h to:

`
enum
  {
    START_BYTE = 0x0F,
    END_BYTE0 = 4,
    END_BYTE1 = 20,
    END_BYTE2 = 36,
    END_BYTE3 = 52
  };
`

and we also changed line 111 of rc_sbus.cpp to 

`
if (byte == START_BYTE && (prev_byte_ == END_BYTE0 || prev_byte_ == END_BYTE1 || prev_byte_ == END_BYTE2 || prev_byte_ == END_BYTE3))
`

and finally we changed line 125 of rc_sbus.cpp to

`
if ((sbus_union_.frame.endByte == END_BYTE0 || sbus_union_.frame.endByte == END_BYTE1 || sbus_union_.frame.endByte == END_BYTE2 || sbus_union_.frame.endByte == END_BYTE3) && sbus_union_.frame.startByte == START_BYTE)
`

Let me know if you have any questions. Thanks!

BillThePlatypus, 2020-02-05:
I've heard some about the SBUS2 protocol that is now supported by some Futaba receivers. Do you know if you are using SBUS or SBUS2?

ccackam, 2020-02-06:
Our receiver has the capability to do both. The original ROSFlight code was not working for us for either option (SBUS or SBUS2). The above code is for SBUS. We have not confirmed that it works with SBUS2.

dpkoch, 2020-03-17:
We're bumping this from v1.3.0 and targeting for v1.3.1 now